plmilord/Hass.io-custom-component-spaclient #1: Hangs installing

14Data, 2020-06-01:
I've completed stages 1 & 2. 

Running Home Assistant 0.108.5

Restart doesn't work, so I've clicked Check Configuration. Starts spinning and never stops. I've left this for an hour or so. I'm guessing this is where it would be installing the crc8 module, or is it hanging here because it's no present. 

I've checked and my BWA app still connects ok. 

Any idea what the issue is and how to resolve it? 

[image]

plmilord, 2020-06-02:
Some tips…

1- Remove `\custom_components\spaclient`
2- Update your HA to the latest version (actually, we're at 0.110.4)
3- Copy `\spaclient` (with all 7 files) in `\custom_components`
4- Edit properly `configuration.yaml`
5- Restart HA
6- ...wait 5 mins... Your HA will automatically install the crc8 package. Give a look at https://pypi.org/project/crc8/ if you are not blocked somewhere!
7- Restart HA
8- You are supposed to see newly created cards in Lovelace interface!

On which device you install HA?
You should not `Local Connect` your spa with the mobile Spa App in parallel… It interferes! `Cloud Connect` should be fine!
Give a try also on a VM… I tried it this Weekend and it worked flawless for me!

plmilord, 2020-06-10:
@14Data, did you make it work for you?

closed 2020-06-16 by plmilord