plmilord/Hass.io-custom-component-spaclient #2: Problems with status reporting

thomasja27, 2020-06-04:
Hi!

Thanks for the job you have done with this one.

Have a problem.
After HA-restart it does report temp and status, and i can control the spa. 
But after a while it stops working. 

When I try to change the setpoint it does not work on the first try, the setpoint temperature bounces back to the value it was at before.
After 2 attempts it works and it reports temperature and status again. 
It works for a while, then it stops working again. When it stops I have to re-adjust the setpoint temperature twice to make it start reporting again.

Is this a problem you've seen before?

My config:
spaclient:
    spa_ip: 192.168.2.6 # Spa IP-adress, Required
    nb_toggle: 1          # 1 or 2 toggle to action the pumps (default = 1), Optional
    scan_interval: 1      # Poll the devices every x seconds (default = 1), Optional

My spa controller:
Balboa BP2100 with 50350 wifi-module

Best regards
Thomas Jakobsen

jaredgudnason, 2020-06-04:
I have a spa, and a similar situation.  I think it's a problem with how their wifi module works.  Anyway, i have a work around that's been working for me.  It's hacky, but works great.  my spa entity in HA has a pump3, but my spa does not.  as you noticed, the first interaction with the spa does kind of a 'wakeup', so i simply have a 5 minute timer, on which i turn pump3 on.  That causes a data refresh to be read, and subsequent commands to work fine.  and even when things are responding properly, does no harm, as there is no pump to turn on anyway.

plmilord, 2020-06-04:
Thanks @thomasja27 and @jaredgudnason for all that feedback!

Interesting, it's the first time I've heard of it! What is the version of your bwa module? You should have something like this: S/N 50350-0_  (where "-0_" is the version). Mine is a bit old… It look like S/N 50350***13081900xx. It sometime stop responding! To resume communication with it, while I notice it with a ping platform, my only way was to toggle a z-wave switch to drop the power momentarily and toggle the first pump once! I know that version 06 or newer can recover when it stops responding.

My assumption about this S/N 50350***13081900xx:
  ***=version (mine is really ***)
  13=yy
  08=mm
  19=dd
  00xx=incremental number during that production day

I will try to add a routine inside this custom component to keep alive the communication channel with the bwa module. Is a fictiv toggle every 5 minutes sufficient?

jaredgudnason, 2020-06-04:
couple observations.  the wifi only really talks to one client at a time.. so if you try to connect with your phone using the app, that will screw it up, but we assume everyone knows that one..
as far as keepalive, etc.  i found the spa generally stopped responding about every 30 minutes, unless i gave it a kick of some sort (issuing a command).  It did GENERALLY seem to correspond with when the pump was kicking in every thirty minutes for circ, but not 100% all the time.  and sometimes it would drop off sooner as well.  (i just setup a display to watch the spa clock time vs my time, so it was obvious when it stopped reporting).
I initially set the timer for 20 minutes, but was getting some larger gaps even with that.  set it down to 5, and it's been "stable" for the last couple weeks anyway.
now.. the other question is.. is it a huge big deal anyway.. not really.  I was just monitoring the pump / heat usage compared to my house power consumption, so wanted to make sure it was staying as real-time as possible.  but in the general scheme of things, every 15-20 minutes probably fine.
What i haven't done yet, but will do so, (so perhaps a little more sensitive to non-responsive), planning on making an alarm when i dont have any updates from the spa for 15+ minutes, as that could indicate a power outage / breaker tripped.

jaredgudnason, 2020-06-04:
and to your question, i have the most recent wifi module version (06).  I think i did observe a wifi failure on the module one day, noticed i had not telemetry for about 15 minutes, checked my AP and it wasn't connected.  but it did self-correct, so i think that was the "every hour reset" that balboa worked into the newer version.

plmilord, 2020-06-04:
Thanks for this sharing of all these observations! I also created automations to be informed when something goes wrong!

Here a view of my Spa in HA with automations listed!

[image]

thomasja27, 2020-06-05:
I think I have the latest version of the WIFI module. It was purchased with the SPA in 2019. I experience the same as jaredgudnason, it works approx. 30 minutes before it stops reporting.

How do you set up automation to monitor communication?
I'm brand new to Home Assistant, so I'm not quite sure how to set things up yet.

plmilord, 2020-06-05:
Regarding my automations... Her is my code:

First, edit `configuration.yaml` to creat a binary sensors:
```
# Binary sensors
binary_sensor:
  - platform: ping
    name: BalboaWiFi
    host: your_spa_ip
    count: 2
    scan_interval: 60
```

Second, edit `automations.yaml` manually or in Lovelace interface to create thoses automations:
```
- alias: Reboot spa WiFi if down for >2min
  trigger:
  - entity_id: binary_sensor.balboawifi
    for: 00:02:00
    platform: state
    to: 'off'
  - entity_id: binary_sensor.balboawifi
    for: 00:05:00
    platform: state
    to: 'off'
  - entity_id: binary_sensor.balboawifi
    for: 00:10:00
    platform: state
    to: 'off'
  action:
  - data:
      entity_id: switch.fortrezz_mimolite_switch
    service: switch.toggle
  - delay: 00:00:30
  - data:
      entity_id: switch.spa_pump_1
    service: switch.toggle

- alias: Notify on spa temperature <90°F
  trigger:
  - below: '90'
    entity_id: climate.spa_temperature
    for: 00:02:00
    platform: numeric_state
    value_template: '{{ state_attr(''climate.spa_temperature'',''current_temperature'')
      }}'
  action:
  - data:
      message: Spa temperature <90°F... Potential leak!
      title: SpaClient
    service: notify.family_distribution_list

- alias: Notify on spa down >1h
  trigger:
  - entity_id: binary_sensor.balboawifi
    for: 01:00:00
    platform: state
    to: 'off'
  condition: []
  action:
  - data:
      message: The spa has not responded for more than an hour!
      title: SpaClient
    service: notify.family_distribution_list
```

thomasja27, 2020-06-08:
Thanks for the code :) Working perfect

thomasja27, 2020-06-09:
There is something strange about temperature reporting. Suddenly it shows 127.5 degrees Celsius. Then it returns to the correct temperature after a while. In the Balboa app, it shows the correct temperature.
[image]
[image]

plmilord, 2020-06-10:
@thomasja27, thank you for this information… I will give it a look!

In the meanwhile, @jaredgudnason, @thomasja27, have a look to the latest files… I've implement two things:
- Keep alive routine
- Automatic time synch

As I introduced a new requirement, you will need to reboot your host.

Let me know if the "App" keeps the communication link alive for you! I toggle a config request each minute to do so… Transparent for the spa!

Have fun!

thomasja27, 2020-06-15:
Polling works very well! I have no automation active to trigger pump 3 now and it updates data just fine :)
Thank you so much!

Time synchronization works, but it seems to change from 24 hours to 12 hours clock. No big problem, but here in Norway we use 24 hour clock. Is there anything I can change in the code so it doesn't end up at 12 hours?

plmilord, 2020-06-16:
@thomasja27, happy that this routine work well for you!

For the time display issue, I know where is the bug... I have to figure out how to set it right! I will work on this bug next weekend!

plmilord, 2020-06-20:
@thomasja27, both repported issues are now adressed!

Current temperature will display nothing while the spa ctrl displays --- ;
Time sync is now take care of what's of preferred display, 12h vs 24h.

...have a look to the latest files!

Let me know results on your side!

thomasja27, 2020-06-21:
Works perfectly :) 
Now it shows no temperature in Home Assistant when the tube shows ---. 
Time sync also doesn't change to 12 hours anymore when I've set it to 24 hours in the app.
Thank you so much for your work!

closed 2020-06-21 by plmilord