plmilord/Hass.io-custom-component-spaclient #4: error loop

jaredgudnason, 2020-08-04:
Hi,
so i had been running the old code (pre last update) for quite some time, but it looks like there was some error condition causing the main HA thread to crash.. so I updated to the most recent code today and have been monitoring.  running into an interesting error condition now (last 30 minutes or so)...  looks like spa may have had a blip / wifi out.  ( i think my wifi module in spa is offline, causing this..)
anyway, i'm tailing my home-assistant.log file, and have an error loop being generated as follows: (pasted below)

the set_time procedure is running into this chunk of code from spaclient.py:
`        try:
            spaclient.s.send(message)
        except IOError as e:
            spaclient.s = spaclient.reconnect_socket()
            spaclient.s.send(message)
`
errors on the first send, hits the except loop, and errors again.  is there a way to change this to something like disposing of the entire object / recreating, or perhaps re-scheduling the timer event to occur again in 5 minutes or something like that?

(also, i had the cover off my tub for the first time in a while yesterday, god only knows.. may have bumped the wire / loose  connection.  I'm going to let this run for the next hour and see if the spa resets the module / self corrects, but it's a very full log in the mean time, generating this error every second.)

`BlockingIOError: [Errno 11] Resource temporarily unavailable
2020-08-04 13:22:32 INFO (MainThread) [schedule] Running job Every 1 hour at 00:05:00 do set_current_time() (last run: 2020-08-04 12:05:00, next run: 2020-08-04 13:05:00)
2020-08-04 13:22:32 ERROR (MainThread) [homeassistant.helpers.entity] Update for climate.spa_temperature fails
Traceback (most recent call last):
  File "/config/custom_components/spaclient/spaclient.py", line 198, in send_message
    spaclient.s.send(message)
BlockingIOError: [Errno 11] Resource temporarily unavailable

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 272, in async_update_ha_state
    await self.async_device_update()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 463, in async_device_update
    await self.async_update()  # type: ignore
  File "/config/custom_components/spaclient/climate.py", line 100, in async_update
    self._spa.read_all_msg()
  File "/config/custom_components/spaclient/spaclient.py", line 185, in read_all_msg
    schedule.run_pending()
  File "/usr/local/lib/python3.8/site-packages/schedule/__init__.py", line 563, in run_pending
    default_scheduler.run_pending()
  File "/usr/local/lib/python3.8/site-packages/schedule/__init__.py", line 94, in run_pending
    self._run_job(job)
  File "/usr/local/lib/python3.8/site-packages/schedule/__init__.py", line 147, in _run_job
    ret = job.run()
  File "/usr/local/lib/python3.8/site-packages/schedule/__init__.py", line 466, in run
    ret = self.job_func()
  File "/config/custom_components/spaclient/spaclient.py", line 221, in set_current_time
    self.send_message(b'\x0a\xbf\x21', bytes([now.hour]) + bytes([now.minute]))
  File "/config/custom_components/spaclient/spaclient.py", line 201, in send_message
    spaclient.s.send(message)
BlockingIOError: [Errno 11] Resource temporarily unavailable
2020-08-04 13:22:32 INFO (MainThread) [schedule] Running job Every 1 hour at 00:05:00 do set_current_time() (last run: 2020-08-04 12:05:00, next run: 2020-08-04 13:05:00)
2020-08-04 13:22:32 ERROR (MainThread) [homeassistant.helpers.entity] Update for light.spa_light fails
Traceback (most recent call last):
  File "/config/custom_components/spaclient/spaclient.py", line 198, in send_message
    spaclient.s.send(message)
BlockingIOError: [Errno 11] Resource temporarily unavailable

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 272, in async_update_ha_state
    await self.async_device_update()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 463, in async_device_update
    await self.async_update()  # type: ignore
  File "/config/custom_components/spaclient/light.py", line 49, in async_update
    self._spa.read_all_msg()
  File "/config/custom_components/spaclient/spaclient.py", line 185, in read_all_msg
    schedule.run_pending()
  File "/usr/local/lib/python3.8/site-packages/schedule/__init__.py", line 563, in run_pending
    default_scheduler.run_pending()
  File "/usr/local/lib/python3.8/site-packages/schedule/__init__.py", line 94, in run_pending
    self._run_job(job)
  File "/usr/local/lib/python3.8/site-packages/schedule/__init__.py", line 147, in _run_job
    ret = job.run()
  File "/usr/local/lib/python3.8/site-packages/schedule/__init__.py", line 466, in run
    ret = self.job_func()
  File "/config/custom_components/spaclient/spaclient.py", line 221, in set_current_time
    self.send_message(b'\x0a\xbf\x21', bytes([now.hour]) + bytes([now.minute]))
  File "/config/custom_components/spaclient/spaclient.py", line 201, in send_message
    spaclient.s.send(message)
BlockingIOError: [Errno 11] Resource temporarily unavailable
`

jaredgudnason, 2020-08-04:
the good news though.. the error control in the most recent version is good enough that my main HA thread is no longer crashing.  so that's a great improvment.

jaredgudnason, 2020-08-04:
ahh.. it did finally recover itself  (assuming spa wifi module reset itself)  only took 30+ minutes.  log file looks normal again, getting updates in HA, and all is good.
so i guess it might not be a huge issue really, would be nice to perhaps knock the frequency down when hitting the 11 - unavailable, but if i'm not actively watching the logs, i suppose i wouldn't know the difference.

jaredgudnason, 2020-08-04:
oh one other thing i did (unrelated).. i was using the pump attributes to monitor the spa prior to the update, and i noticed you removed the majority of those attributes in the update.  While i certainly could have done some work in my UI to fix, i just took the cheater way and added the attributes back in switch.py:  (although i'm not sure if you were exposing the time attribute anywhere else)

`    @property
    def device_state_attributes(self):
        """Return the state attributes of the device."""
        attrs = {}
#        attrs['Spa Pump ' + str(self._pump_num)] = self._spa.get_pump(self._pump_num)

        attrs["Current Temp"] = self._spa.get_current_temp()
        attrs["Time"] = self._spa.get_current_time()
        attrs["Set Temp"] = self._spa.get_set_temp()
        attrs["Pump 1"] = self._spa.get_pump(1)
        attrs["Pump 2"] = self._spa.get_pump(2)
        attrs["Pump 3"] = self._spa.get_pump(3)
        attrs["Temp Range"] = self._spa.get_temp_range()

        return attrs
`

plmilord, 2020-08-05:
@jaredgudnason , thanks for all the feedback ... It looks like when updating this custom component and your HA something hangs somewhere! Glad that everything returned to a normal state!

Yes, in my last version I did some code cleanup and got the idea to create sensors to display these attributes. Since then, I'm working to create a complete integration for HA, and I will try to display those sensors in the Spa Client integration page. For now, I'm coding a little in the dark because my spa is de-energized since two months (constructions in my backyard). My target is to publish a complete new design of this app early this fall!

natekspencer, 2020-09-10:
@plmilord, I've created a custom component based on the work done by you and @garbled1. I took the python library that was created and filled in some missing gaps about module/system/device configuration. My version should only create the devices if they are part of the system (like number of pumps/lights/blower/mister/etc). Feel free to look at it and collaborate as desired!

Here's the custom component: https://github.com/natekspencer/spaclient_hacs
And here's the python code: https://github.com/natekspencer/pybalboa

plmilord, 2020-09-14:
@natekspencer, great! I will give it a try when I will get my spa back! It looks like with your version we're pretty close to a full official Home Assistant release (that was my goal). @garbled1 's code is a good base (better than mine in my opinion!). I will participate as much as I can to your project to bring a full relaible alternative solution to that bwa spa App.

mattssa82, 2021-01-11:
@plmilord @natekspencer @ hi is there a version how is working great? with a separate temp sensor for logging

natekspencer, 2021-01-11:
@mattssa82 I've put all my work into https://github.com/garbled1/balboa_homeassistan and it's been working great for me. The temperature sensor is on the climate device, so you can trigger anything you want off of that or even create a template_sensor if you want to track it differently.

plmilord, 2021-06-05:
Please take a look at the new version 2.0 ... this will fix a lot of bugs. Inspired by the work of @natekspencer and @garbled1...

No more "schedule" add-on dependencies... No dependencies at all!

Regards,

closed 2021-06-05 by plmilord