plmilord/Hass.io-custom-component-spaclient #6: Light not showing in home assistant

B-Hartley, 2021-09-04:
Hi,

I've been using.......
https://github.com/garbled1/balboa_homeassistan

But recently found your component, so I'm giving it a go.
My lights have not appeared in your component, but they did in the other one.
Is there anything I can do to make them work? or to help you get them to work?

Thanks,
Bruce.

plmilord, 2021-09-04:
It looks weird...

Could you copy and paste the `<< Configuration variables >>` section found in your `home-assistant.log`?

Have you tried reinstalling the custom component?

B-Hartley, 2021-09-04:
I did a home assistant restart and it appeared. 

Thanks.

B-Hartley, 2021-09-05:
I notice your component displays decimal temps such as 36.9
The previous one and the tub itself only shows .5 degrees. 
Are you provided with accurate temps or is it some kind of unit conversion issue ?

closed 2021-09-05 by B-Hartley