plmilord/Hass.io-custom-component-spaclient #7: Pump switches bounce when switched in HA

B-Hartley, 2021-09-05:
Hi,

When I try and switch on or off a pump using the home assistant switches they "bounce".
So if I switch a pump on, it physically switches on for half a second and then goes immediately off.

The switch on screen returns to off at the next update.

Can't find any debug of when I flip the switch, but my settings info is........

2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] 
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] ======================
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] << socket variables >>
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] ======================
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] self.is_connected = True
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] self.l       = <unlocked _thread.lock object at 0x7fab6d960450>
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] self.s       = <socket.socket fd=23, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.11.*HomeAssistant*', 55470), raddr=('192.168.21.169', 4257)>
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] self.host_ip = 192.168.21.*SPA*
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] 
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] =============================
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] << Status update variables >>
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] =============================
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] self.hold_mode    = False
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] self.priming      = False
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] self.current_temp = 100.5
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] self.hour         = 10
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] self.minute       = 28
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] self.heat_mode    = Ready
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] self.temp_scale   = Celsius
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] self.filter_mode  = 1
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] self.time_scale   = 24 Hr
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] self.heating      = 0
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] self.temp_range   = High
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] self.pump1        = Off
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] self.pump2        = Off
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] self.pump3        = Off
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] self.pump4        = Off
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] self.pump5        = Off
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] self.pump6        = Off
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] self.circ_pump    = 2
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] self.blower       = 0
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] self.light1       = False
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] self.light2       = False
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] self.mister       = 0
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] self.aux1         = 0
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] self.aux2         = 0
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] self.set_temp     = 87.0
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] 
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] ===========================
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] << Information variables >>
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] ===========================
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] self.information_loaded = True
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] self.model_name     = BP200G2
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] self.cfg_sig        = 89ca476c
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] self.sw_vers        = 52.0
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] self.setup          = 1
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] self.ssid           = M100_235 V52.0
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] self.heater_voltage = Unknown
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] self.heater_type    = Unknown
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] self.dip_switch     = 0000010000000000
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] 
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] =============================
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] << Configuration variables >>
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] =============================
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] self.configuration_loaded = True
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] self.pump_array      = [1, 1, 0, 0, 0, 0]
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] self.light_array     = [1, 0]
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] self.circ_pump_array = [1]
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] self.blower_array    = [0]
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] self.mister_array    = [0]
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] self.aux_array       = [0, 0]
[... 7 lines cut ...]
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] 
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] =============================
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] << Filter cycles variables >>
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] =============================
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] self.filter_cycles_loaded = True
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] self.filter1_hour             = 10
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] self.filter1_minute           = 0
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] self.filter1_duration_hours   = 2
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] self.filter1_duration_minutes = 0
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] self.filter2_enabled          = 0
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] self.filter2_hour             = 16
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] self.filter2_minute           = 0
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] self.filter2_duration_hours   = 2
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] self.filter2_duration_minutes = 0
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] 
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] ======================================
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] << Additional information variables >>
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] ======================================
[... cut ...]
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] self.low_range_min  = 50
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] self.low_range_max  = 99
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] self.high_range_min = 80
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] self.high_range_max = 104
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const] self.nb_of_pumps    = 2
2021-09-05 10:28:11 INFO (MainThread) [custom_components.spaclient.const]

plmilord, 2021-09-05:
Please replace `spaclient.py` and `switch.py` with the newly updated files.

... This should solve your pump problem!

Let me know if this is working correctly for you ...

====

In addition, in your configuration presented here above, I saw that you have a "circulation pump"... Is the component shows properly the device? Possible status are only "On" and "Off"? Let me know...!

Thx!

B-Hartley, 2021-09-05:
That works perfectly, thank you very much !!

And yes, binary_sensor.spa_circ_pump display 'on' when the circ pump is running.

Thank you

closed 2021-09-05 by B-Hartley