plmilord/Hass.io-custom-component-spaclient #8: Round temperature to nearest 0.5 when displaying in Celsius

B-Hartley, 2021-09-05:
Hi,

From looking at the info in homeassistant log, it looks like it provides you the temp in F even when it tells you that display is C.
The conversion seems to have a slight error compared with what it would display on the device.
Could you round to the nearest 0.5C to make it match?

Thanks,
Bruce.

plmilord, 2021-09-06:
Could you try the v2.1 package...

This should fix the Celsius to Fahrenheit issue. Now bumpless regardless of your HA setting or spa. And rounded to the nearest 0.5C while in Celsius!

Let me know if this work for you... Also test a second time your pump (I adjusted a small thing).

B-Hartley, 2021-09-06:
All looks good. Thanks.

closed 2021-09-06 by B-Hartley