plmilord/Hass.io-custom-component-spaclient #11: I need some help from my fellow Balboa Owners.

tonyfitzs, 2021-11-11:
Hi there, 

I am reaching out the my fellow Balboa Spa controller owners because I need some help with something. This is not directly related to the BWA client. Its more about the controller its self. 
A few days ago one of the pumps in my spa took a power spike that fused the motor. This also caused a huge current spike that resulted in an arc at the point where the power goes into the controller. 

I’m confident the controller is still able to be repaired, one thing that is great about the Balboa controller is they are almost bullet proof. Mind you I have never tried to shoot it, although I have been tempted. 

Anyway, I have a Balboa Revolution controller (BP1600) and I need to either get a copy of a circuit diagram or else just a picture of the board where the neutral power goes into the controller. I have attached a couple of photos of what I am talking about, I need to know if the neutral line or the active line connects to U11, as you can imaging this is a very significant difference. 

Thanks in advance, and sorry for the off-topic request, but I have run out of other options.

[image]
[image]

plmilord, 2021-11-12:
I can't help on this one.  My spa ctrl doesn't look like this!

tonyfitzs, 2021-11-12:
Ok thanks anyway.

ric866, 2021-11-17:
Hi Both,

Firstly Pierre, I am most grateful for your work with the BWA WiFi module. It's nice to have such a problematic device under at least some kind of monitoring!

Tony, I'm not a qualified person (just to put that out there), but I've spent more time than I wanted looking inside hot tubs. 
You might find this useful : [BWG Service Manual](https://www.balboawatergroup.com/getdoc.cfm?id=1611) 
Page 27 on the right shows that link, and by the looks of it, it goes past the bottom terminal, to the middle one.

Hope that helps (even if a little). 

Kind regards

Ric

tonyfitzs, 2021-11-17:
Thanks Ric,

This is very helpful.

enkrypt3d, 2021-11-25:
Tony it may just be easier to buy the BP7 spa pack and start fresh... it's not too expensive.

plmilord, 2021-11-28:
Sounds it has been answered by @ ric 866 ... I close this issue! Thanks for this great teamwork!

closed 2021-11-28 by plmilord