plmilord/Hass.io-custom-component-spaclient #12: threw error on integration activation and then nothing.

anon-noname, 2021-11-22:
I posted a more detailed description at Home Assistant forum. 

Directions are simple enough but don’t correspond or work for me. directions say "/config/custom_components ". I did not have one. I did have an “/opt/hassio/homeassistant/custom_components” which has two folders - hacs and nodered.
adding "spaclient " folder there did not work even after restart. I made folder and moved to /opt/hassio/homeassistant/config and rebooted. A spa icon appeared in the list when I went to add integration but when I clicked on it there was an error message which I can’t recall details of . Since I hadn’t matched the path in the instructions, I moved folder into /opt/hassio/homeassistant/config/custom_components and tried again. No help. Now the icon is gone and doesn’t re-appear even if I put the folder back (and reboot) where it began to work! It was also not clear where the associated images folder is to go.

Thanks so much for putting this together! I have a 2 jet pump unit with wifi, lights, ozone, and A/V.  I am willing to help test/prove as time permits. Close second priority for me is to get HA moved onto RPI4 8G with 250G ssd to remove memory constraints and hopefully boost responsiveness.

enkrypt3d, 2021-11-25:
Make sure everything is in /usr/share/hassio/homeassistant/custom_components/spaclient including images and translations then restart HA. then refresh your integrations and it should show up

plmilord, 2021-11-27:
@anon-noname, were you able to manage this spaclient component?

When everything works for you, you can help develop this component... Here are the instructions for sharing your spa setup:

To help develop this component, can you share your spa setup? See the printout in `home-assistant.log` ... in particular the sections `Information variables` and `Configuration variables`.

...To make this append, you should activate `logger`. To do so, add this in your `configuration.yaml`:

```
# Enable the logger integration
logger:
    default: error
    logs:
        custom_components.spaclient: info
```

...restart Home Assistant, the `home-assistant.log` will show information regarding your setup.

anon-noname, 2021-11-28:
thanks for looking at this. I am getting close. I decided to give the RPI another whack. Another full backup, download and burn up to date image and restore. There are still some things that aren't up to speed but they are out of scope. The Icon now shows up when I go to add integration but I don't know where to find the data to enter for <host> in the window that comes up with title "Connect to the bwa Wi-Fi Module"!

plmilord, 2021-11-28:
All you need to do is enter the IP of your spa. Should look something like this: 192.168.?.?

Take a look at your router's IP table to retrieve it! And a good idea is also to fix this IP address in your router!

anon-noname, 2021-12-04:
After an accidental command to re-install HA on this system (should have been on replacement system) integration showed up again and I was able to configure, add controls and indicators in lovelace and send commands (that worked) all in about 15 minutes. Thanks again! Once I have transitioned to replacement system which is RPI4 8G I will provide any data and test any thing I can but it probably won't happen 'till Christmas break. Too many irons in fire.

closed 2021-12-04 by anon-noname