plmilord/Hass.io-custom-component-spaclient #14: Issue with Blower status

sebitr, 2021-11-27:
Hi,

First of all thanks a lot for this awesome integration. It works way better than the balboa app.

I have only one issue:
The blower sensor stays on all the time in home assistant.
However the button triggers the on/off action on the spa itself.
So I guess it’s just a matter of status.
I can give you access to my HA if it helps. And thanks again :)

plmilord, 2021-11-27:
Hello @sebitr,

I have adjusted a few things in the code to help you get this Blower working... As you are the first here to own such a component, you will help me validate!

Please replace the original 'spaclient.py' and 'switch.py' files with the ones uploaded [here](https://github.com/plmilord/Hass.io-custom-component-spaclient/files/7612452/spaclient.zip)... if that works for you, I will upload them to the main project.

Thank you for your follow-up!

Also...

To help develop this component, can you share your spa setup? See the printout in `home-assistant.log` ... in particular the sections `Information variables` and `Configuration variables`.

...To make this append, you should activate `logger`. To do so, add this in your `configuration.yaml`:

```
# Enable the logger integration
logger:
    default: error
    logs:
        custom_components.spaclient: info
```

...restart Home Assistant, the `home-assistant.log` will show information regarding your setup.

sebitr, 2021-11-28:
Hello @plmilord,

Thanks a lot, it works perfectly. You are genius.

Here is the full log:
`2021-11-28 10:09:43 INFO (MainThread) [custom_components.spaclient.const] ======================
2021-11-28 10:09:43 INFO (MainThread) [custom_components.spaclient.const] << socket variables >>
2021-11-28 10:09:43 INFO (MainThread) [custom_components.spaclient.const] ======================
2021-11-28 10:09:43 INFO (MainThread) [custom_components.spaclient.const] self.is_connected = True
2021-11-28 10:09:43 INFO (MainThread) [custom_components.spaclient.const] self.l       = <unlocked _thread.lock object at 0x705fedd0>
2021-11-28 10:09:43 INFO (MainThread) [custom_components.spaclient.const] self.s       = <socket.socket fd=21, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.x.x', 40142), raddr=('192.168.x.x', 4257)>
2021-11-28 10:09:43 INFO (MainThread) [custom_components.spaclient.const] self.host_ip = 192.168.x.x
2021-11-28 10:09:43 INFO (MainThread) [custom_components.spaclient.const] 
2021-11-28 10:09:43 INFO (MainThread) [custom_components.spaclient.const] =============================
2021-11-28 10:09:43 INFO (MainThread) [custom_components.spaclient.const] << Status update variables >>
2021-11-28 10:09:43 INFO (MainThread) [custom_components.spaclient.const] =============================
2021-11-28 10:09:43 INFO (MainThread) [custom_components.spaclient.const] self.hold_mode    = False
2021-11-28 10:09:43 INFO (MainThread) [custom_components.spaclient.const] self.priming      = True
2021-11-28 10:09:43 INFO (MainThread) [custom_components.spaclient.const] self.current_temp = None
2021-11-28 10:09:43 INFO (MainThread) [custom_components.spaclient.const] self.hour         = 10
2021-11-28 10:09:43 INFO (MainThread) [custom_components.spaclient.const] self.minute       = 9
2021-11-28 10:09:43 INFO (MainThread) [custom_components.spaclient.const] self.heat_mode    = Rest
2021-11-28 10:09:43 INFO (MainThread) [custom_components.spaclient.const] self.temp_scale   = Celsius
2021-11-28 10:09:43 INFO (MainThread) [custom_components.spaclient.const] self.filter_mode  = 0
2021-11-28 10:09:43 INFO (MainThread) [custom_components.spaclient.const] self.time_scale   = 24 Hr
2021-11-28 10:09:43 INFO (MainThread) [custom_components.spaclient.const] self.heating      = 0
2021-11-28 10:09:43 INFO (MainThread) [custom_components.spaclient.const] self.temp_range   = High
2021-11-28 10:09:43 INFO (MainThread) [custom_components.spaclient.const] self.pump1        = Off
2021-11-28 10:09:43 INFO (MainThread) [custom_components.spaclient.const] self.pump2        = Off
2021-11-28 10:09:43 INFO (MainThread) [custom_components.spaclient.const] self.pump3        = Off
2021-11-28 10:09:43 INFO (MainThread) [custom_components.spaclient.const] self.pump4        = Off
2021-11-28 10:09:43 INFO (MainThread) [custom_components.spaclient.const] self.pump5        = Off
2021-11-28 10:09:43 INFO (MainThread) [custom_components.spaclient.const] self.pump6        = Off
2021-11-28 10:09:43 INFO (MainThread) [custom_components.spaclient.const] self.circ_pump    = Off
2021-11-28 10:09:43 INFO (MainThread) [custom_components.spaclient.const] self.blower       = Off
2021-11-28 10:09:43 INFO (MainThread) [custom_components.spaclient.const] self.light1       = False
2021-11-28 10:09:43 INFO (MainThread) [custom_components.spaclient.const] self.light2       = False
2021-11-28 10:09:43 INFO (MainThread) [custom_components.spaclient.const] self.mister       = 0
2021-11-28 10:09:43 INFO (MainThread) [custom_components.spaclient.const] self.aux1         = 0
2021-11-28 10:09:43 INFO (MainThread) [custom_components.spaclient.const] self.aux2         = 0
2021-11-28 10:09:43 INFO (MainThread) [custom_components.spaclient.const] self.set_temp     = 76
2021-11-28 10:09:43 INFO (MainThread) [custom_components.spaclient.const] 
2021-11-28 10:09:43 INFO (MainThread) [custom_components.spaclient.const] ===========================
2021-11-28 10:09:43 INFO (MainThread) [custom_components.spaclient.const] << Information variables >>
2021-11-28 10:09:43 INFO (MainThread) [custom_components.spaclient.const] ===========================
2021-11-28 10:09:43 INFO (MainThread) [custom_components.spaclient.const] self.information_loaded = True
2021-11-28 10:09:43 INFO (MainThread) [custom_components.spaclient.const] self.model_name     = BP6013G2
2021-11-28 10:09:43 INFO (MainThread) [custom_components.spaclient.const] self.cfg_sig        = f71fe5eb
2021-11-28 10:09:43 INFO (MainThread) [custom_components.spaclient.const] self.sw_vers        = 43.0
2021-11-28 10:09:43 INFO (MainThread) [custom_components.spaclient.const] self.setup          = 6
2021-11-28 10:09:43 INFO (MainThread) [custom_components.spaclient.const] self.ssid           = M100_226 V43.0
2021-11-28 10:09:43 INFO (MainThread) [custom_components.spaclient.const] self.heater_voltage = Unknown
2021-11-28 10:09:43 INFO (MainThread) [custom_components.spaclient.const] self.heater_type    = Standard
2021-11-28 10:09:43 INFO (MainThread) [custom_components.spaclient.const] self.dip_switch     = 0000000000000000
2021-11-28 10:09:43 INFO (MainThread) [custom_components.spaclient.const] 
2021-11-28 10:09:43 INFO (MainThread) [custom_components.spaclient.const] =============================
2021-11-28 10:09:43 INFO (MainThread) [custom_components.spaclient.const] << Configuration variables >>
[... cut ...]
2021-11-28 10:09:43 INFO (MainThread) [custom_components.spaclient.const] self.additional_information_loaded = True
2021-11-28 10:09:43 INFO (MainThread) [custom_components.spaclient.const] self.low_range_min  = 50
2021-11-28 10:09:43 INFO (MainThread) [custom_components.spaclient.const] self.low_range_max  = 99
2021-11-28 10:09:43 INFO (MainThread) [custom_components.spaclient.const] self.high_range_min = 80
2021-11-28 10:09:43 INFO (MainThread) [custom_components.spaclient.const] self.high_range_max = 104
2021-11-28 10:09:43 INFO (MainThread) [custom_components.spaclient.const] self.nb_of_pumps    = 2`

plmilord, 2021-11-28:
Thanks for your spa informations...

To consolidate my knowledge... Questions:
- `Circulation Pump` states are only On/Off within your spa control pannel -or- bwa App (we can't control it manually, no speed, etc.)?
- `Blower` states are only On/Off within your spa control pannel -or- bwa App (no speed, etc.)?
- Whate is your heater voltage? (self.heater_voltage = Unknown)

...For 'Circulation Pump', in Spa Client v2.1, I manage states with binaries (0-1)... In the files shared with you, I changed my method to improve understanding in the code... Did you see a difference? Still working good?

sebitr, 2021-11-29:
Hi @plmilord 

- Circulation Pump states are only On/Off within your spa control pannel or bwa App (we can't control it manually, no speed, etc.)?  --> Correct. It's not possible to manually control the pump. It's done via the filer cycles 1 and 2 only. There is no speed.
- Blower states are only On/Off within your spa control pannel or bwa App (no speed, etc.)? --> Correct, only on/off.
- What is your heater voltage? (self.heater_voltage = Unknown) --> It's a 3kW Balboa heater. Where can I find the voltage information?

...For 'Circulation Pump', in Spa Client v2.1, I manage states with binaries (0-1)... In the files shared with you, I changed my method to improve understanding in the code... Did you see a difference? Still working good?  --> Not sure it still works. It looks like it's always on now. Even when both filtration cycles are off.

plmilord, 2021-12-04:
Any changes with the "Circulation pump"? So it no longer displays correctly?

plmilord, 2021-12-11:
@sebitr, I assumed that everything's work great! I realesed v2.2 and closed this issue!

closed 2021-12-11 by plmilord