plmilord/Hass.io-custom-component-spaclient #17: Seeing these errors in my logs - anything to worry about?

enkrypt3d, 2022-02-20:
2022-02-19 21:04:11 INFO (MainThread) [homeassistant.components.binary_sensor] Setting up binary_sensor.spaclient
2022-02-19 21:04:11 INFO (MainThread) [homeassistant.components.climate] Setting up climate.spaclient
2022-02-19 21:04:11 INFO (MainThread) [homeassistant.components.light] Setting up light.spaclient
2022-02-19 21:04:11 INFO (MainThread) [homeassistant.components.switch] Setting up switch.spaclient
2022-02-19 21:04:11 WARNING (MainThread) [homeassistant.helpers.entity] Entity binary_sensor.filter_cycle_1_status (<class 'custom_components.spaclient.binary_sensor.FilterCycle'>) implements device_state_attributes. Please report it to the custom component author.
2022-02-19 21:04:11 WARNING (MainThread) [homeassistant.helpers.entity] Entity binary_sensor.filter_cycle_2_status (<class 'custom_components.spaclient.binary_sensor.FilterCycle'>) implements device_state_attributes. Please report it to the custom component author.
2022-02-19 21:04:11 WARNING (MainThread) [homeassistant.helpers.entity] Entity switch.pump_1 (<class 'custom_components.spaclient.switch.SpaPump'>) implements device_state_attributes. Please report it to the custom component author.
2022-02-19 21:04:11 WARNING (MainThread) [homeassistant.helpers.entity] Entity switch.heat_mode (<class 'custom_components.spaclient.switch.HeatMode'>) implements device_state_attributes. Please report it to the custom component author.
2022-02-19 21:04:11 WARNING (MainThread) [homeassistant.helpers.entity] Entity switch.temperature_range (<class 'custom_components.spaclient.switch.TempRange'>) implements device_state_attributes. Please report it to the custom component author.

plmilord, 2022-02-21:
Use the latest version of Spa Client (v2.4)... which was released a month ago to resolve this Home Assistant tweak.

enkrypt3d, 2022-02-21:
Ok updated! I'll let you know how it goes, thanks!

plmilord, 2022-02-21:
If everything's fine on your side, do not hesitate to close this issue! Thank-you!

enkrypt3d, 2022-02-21:
I had to revert to the old version. Maybe I didn't update the right files.
I only grabbed anything that's been changed recently. Do I need to download
the whole thing again?

plmilord, 2022-02-21:
It's better to Uninstall the old version (through the HA web interface), remove what's in /custom_components/spaclient and recreate the /spaclient with the new files... then reconfigure your spa (through the HA web interface)! Piece of mind after that!

enkrypt3d, 2022-02-22:
Reloading the integration with the new code seemed to work! Thanks!

closed 2022-02-22 by enkrypt3d