plmilord/Hass.io-custom-component-spaclient #19: set light color

mclever, 2022-05-01:
Is there any way to set the light color via HA?

I tried the built-in integration, https://github.com/garbled1/balboa_homeassistan, and yours, none seem to offer the light color selector like this
[image]

plmilord, 2022-05-01:
It would be very interesting, but I think there is no way to make it possible. The spa only activates a standard switch to power the LED module. There are no other exchanges with this LED module.

Unless you tell me that you have this option on your control interface on your spa...

mclever, 2022-05-01:
I just have on/off that cycles through the colors on touch screen. I need to have a few leds swapped out soon and I'll poke around then and see how the lights are connected to controller.

plmilord, 2022-05-01:
Same behavior here. It is possible to believe that we can make the lighting system independent of the controller (but will also remain independent of this component).

jaredgudnason, 2022-05-01:
> ID: <plmilord/Hass.io-custom-component-spaclient/issues/19/1114290342@
> github.com>

I looked at this extensively when I got the hot tub a couple years ago.
Depending on your lighting system, some are literally just a power
connection to the controller. It cycles through light settings with power
on/off.  (this is what I have)
The more advanced controller has a control tie in to the tub controller and
the screen. Requires updated firmware on controller and newer screen
interface.
The LEDs themselves are just chained to the control box, treat them like a
single color RGB led strip.
You absolutely could put in your own lighting controller, but it's not
going to have any tie in to the panel on the tub, excepting power on/off if
you use that power line coming from the tub controller.

plmilord, 2022-05-24:
Not an issue... general discussion on the idea of improving the LED lighting.

closed 2022-05-24 by plmilord