plmilord/Hass.io-custom-component-spaclient #20: Entity missing?

enkrypt3d, 2022-05-27:
I used to have an orange graph that showed the entity for when the heater is running.... 

[image]

Now it's completely gone. On the latest version of HA.

plmilord, 2022-05-27:
Everything's fine on my side... Here's my setup:

Home Assistant Core 2022.5.5
Home Assistant Supervisor 2022.05.3
Home Assistant OS 8.1

Sounds like a web browser cache issue!?

Have you tried:
- different web browser?
- clear web browser cache?
- Uninstall/reinstall the component?

Let me know your walkthrough...

plmilord, 2022-05-28:
No need to show the heater heating as there's no heating... your target temperature was 50F and your current temperature was close to 80F, dropping! It also looks like you have disabled the temperature range and heating mode (because the current temperature reading goes to nothing at times... normal behavior of your controller).

enkrypt3d, 2022-06-01:
ok once it cools down I'll run it again and see if the entity shows up... odd that it just disappears though when it's not in use?

plmilord, 2022-06-05:
...and, everything's come back to normal?

enkrypt3d, 2022-06-06:
we can close this for now. I'll re-open it later on this year when it's hot tub season if there is an issue. Thanks for your support!

closed 2022-06-06 by enkrypt3d