plmilord/Hass.io-custom-component-spaclient #21: two pools

bopbrev, 2022-06-04:
I have two pools, and get the second pool created, but this pool only gets one entity, and the other pool loses one entity. Is there anything I need to do to create two pools?

plmilord, 2022-06-05:
The ability to add more than one Spa is still something I need to implement in the code... Should not be so hard. Give me some time and I will come back to you to test such development.

plmilord, 2022-10-21:
I worked on it last summer. I still need time to complete the tests. As soon as I have some time, I'll give it a look. Thank you for your patience!

bopbrev, 2022-10-21:
great, let me know if you want me to test anything

plmilord, 2023-08-22:
I am currently working on another custom component and have implemented it...just need to adapt the code accordingly. Need time, that's the current challenge ;-)

plmilord, 2023-11-02:
Spa Client v2.8 has just been released. This version now allows multiple integrations in the same Home Assistant instance.

Thanks for your patience @bopbrev on this one. Let me know if this works correctly for you... through Discussions (https://github.com/plmilord/Hass.io-custom-component-spaclient/discussions)

closed 2023-11-02 by plmilord