plmilord/Hass.io-custom-component-spaclient #32: Feature: Messages from the Spa

Rmh1978, 2023-06-19:
It would be really nice if there could be entity or attribute stating the messages from the Spa like when the Spa display says:

"Filter change"

plmilord, 2023-08-22:
It's on my to-do list... Since there's some info on these posts, I think it'll be easy. I just need a little time to do it!

plmilord, 2023-11-15:
No... I haven't implemented this yet, but it could be done quite easily...

Please refer to the glossary of what is currently known in the messages transmitted by the Balboa module:

https://github.com/ccutrer/balboa_worldwide_app/wiki#fault-message-codes

I will come back with this in a future version to make the spa status messages known.

plmilord, 2024-07-16:
Spa messages are now implemented in [Spa Client v3.0](https://github.com/plmilord/Hass.io-custom-component-spaclient/releases/tag/v3.0). As stated previously, the implemented messages are those listed [here](https://github.com/ccutrer/balboa_worldwide_app/wiki#fault-message-codes). If you discover other codes, don't hesitate to share them with us!

closed 2024-08-03 by plmilord

Jieper001, 2024-08-03:
Just to show my huge gratitude for the work @plmilord is putting in.  My 2y old spa's touch display is already dead and still enjoying the spa via this great integration.  THANK YOU

evertide, 2024-08-03:
Many thanks….