plmilord/Hass.io-custom-component-spaclient #34: Wifi Module losing connection?

Inrego, 2023-07-05:
I admit, this question is not directly related to this project.

I'm having issues with my WiFi module losing connection for hours at a time. Has anyone else experienced similar issues? I'm suspecting bad coverage of WiFi at the installation, but it should be alright.

plmilord, 2023-07-08:
Do you have more than one AP? Signal strength?

I had similar problems in the past. I approached an AP and made sure to prioritize that AP in my network configuration. As the module gets mixed up on occasion, I also installed a remote controlled contactor on the power supply of the Balboa WiFi module.

Inrego, 2023-07-09:
Yes I do have more than one AP. However, I can see that it only connects to the nearest one.

It might be signal strength, but I wanted to rule out if it was normal behavior before spending money to set up another outdoor AP (our spa is an outdoor swim spa).

I can't believe the app is so bad, that I can't access WiFi settings once it's connected. It gets a new IP address every time it reconnects, so I want to configure network settings to a fixed IP address. For now, I've reserved it to a fixed IP address in my DHCP server, but my experience tells me that's not 100% foolproof

plmilord, 2023-07-09:
The 50350 Wifi module has several revisions... Rev. 6 automatically reconnects after losing connection. Rev. 2, the one I have, the only way to get it back on track is to toggle the power of the module. I'm toggling the power with the binary sensor `bwa Wi-Fi Module` included in this custom component. As we do not fully control the stability of our Wifi, you should limit your module to a single access point in your network configuration.

pnomme, 2023-07-23:
I have two wifi modules that used to disconnect randomly until I installed an outside access point. Now both the devices in my spa/swimspa are in general stable, disconnects occasionally but reconnects automatically.
This is the AP I'm using:
https://eu.store.ui.com/eu/en/collections/unifi-wifi-outdoor-flexible

Inrego, 2023-07-23:
Thanks. I'm using Omada instead, but it's pretty much the same. I've ordered a new outdoor AP to extend range to improve stability.

But so far, stability has already improved by reserving its IP address in DHCP.

closed 2023-08-22 by plmilord