plmilord/Hass.io-custom-component-spaclient #36: filter 1

martinmasera, 2023-08-18:
hello,

thanks for great app. bytheway I found out that filter 1 can not be setup to on and off and that the app do not control the system when filter 1 is running and stoped, but this matter still control by control spa panel.

any help how to implement the on/off button for filter 1 programm and how to get a real value for this filter start time and end time.

plmilord, 2023-08-22:
You can find this information in the `Filter Cycle 1 Status` attributes (Begins, Runs, Ends).

Filter cycle 1 does not have a native option to enable/disable it. If you have this option through your Coast Spas app, we'll have to investigate!

martinmasera, 2023-08-22:
Yes, you are right do not have the option to disable and enable, thanks a lot.

closed 2023-08-24 by plmilord