plmilord/Hass.io-custom-component-spaclient #40: No Module named homeassistant.Util.temperature (conversion utility deprecated)

B-Hartley, 2023-10-28:
Hi,

Running 2023.11.0b1 on OS 11.1

Suspect problems due to:
https://developers.home-assistant.io/blog/2022/09/28/deprecate-conversion-utilities/

Needs changing to:
homeassistant/util/unit_conversion:
TemperatureConverter

```
2023-10-28 16:07:43.969 ERROR (MainThread) [homeassistant.setup] Setup failed for custom integration spaclient: Unable to import component: No module named 'homeassistant.util.temperature'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 215, in _async_setup_component
    component = integration.get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 816, in get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_components/spaclient/__init__.py", line 19, in <module>
    from .spaclient import spaclient
  File "/config/custom_components/spaclient/spaclient.py", line 9, in <module>
    from homeassistant.util.temperature import convert as convert_temperature
ModuleNotFoundError: No module named 'homeassistant.util.temperature'
```

B-Hartley, 2023-10-28:
Not a "proper fix"
But I've done this to make it work bv changing spaclient.py

```
#from homeassistant.const import TEMP_CELSIUS, TEMP_FAHRENHEIT
from homeassistant.const import UnitOfTemperature
#from homeassistant.util.temperature import convert as convert_temperature
from homeassistant.util.unit_conversion import TemperatureConverter
```

```
            #temp = round(convert_temperature(temp, TEMP_FAHRENHEIT, TEMP_CELSIUS) * 2)
            temp = round(TemperatureConverter.convert(
                temp, UnitOfTemperature.FAHRENHEIT, UnitOfTemperature.CELSIUS
            ) * 2)
```

plmilord, 2023-11-02:
@B-Hartley, thanks for that warning...

For `UnitOfTemperature`, I will adjust shortly...

For `convert_temperature` warning, sounds like you run a very old Spa Client version? This was fixed last December in v2.6... with your help!?

plmilord, 2023-11-02:
Spa Client v2.8 has just been released to resolve this obsolescence problem.

If you find other anomalies, don't hesitate!

A big thank you @B-Hartley

closed 2023-11-02 by plmilord