plmilord/Hass.io-custom-component-spaclient #41: Unable to import component: No module named 'homeassistant.util.temperature'

enkrypt3d, 2023-11-02:
Getting this error after upgrading to the latest version 2023.11

Logger: homeassistant.setup
Source: setup.py:215
First occurred: 11:27:51 AM (1 occurrences)
Last logged: 11:27:51 AM

Setup failed for custom integration spaclient: Unable to import component: No module named 'homeassistant.util.temperature'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 215, in _async_setup_component
    component = integration.get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 816, in get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_components/spaclient/__init__.py", line 19, in <module>
    from .spaclient import spaclient
  File "/config/custom_components/spaclient/spaclient.py", line 9, in <module>
    from homeassistant.util.temperature import convert as convert_temperature
ModuleNotFoundError: No module named 'homeassistant.util.temperature'

plmilord, 2023-11-03:
What version of Spa Client are you using? If you just updated to v2.8, I suggest you delete the /spaclient folder inside the /custom_components folder. Restart Home Assistant, then reinstall v2.8.

What is your system infos? Mine:

Core 2023.11.0
Supervisor 2023.10.1
Operating System 11.1
Interface utilisateur 20231030.1

enkrypt3d, 2023-11-03:
I have it installed via HACS. There is nothing in the custom_components
folder for the spa client.

Core 2023.11.0
Supervisor 2023.10.1
Operating System 11.1

plmilord, 2023-11-04:
If there is nothing in your custom components folder, I suggest you copy the /spaclient to /custom_components. Restart Home Assistant and try uninstalling it again. This should work… then restart HA and reinstall Spa Client.

enkrypt3d, 2023-11-04:
After banging my head on the wall it's working again. no idea why the upgrade to v11 caused this. thanks!

closed 2023-11-04 by enkrypt3d