plmilord/Hass.io-custom-component-spaclient #42: Unable to load integrations

IkaSGC, 2023-11-04:
It just... Do you know what it could be? 

"Logger: homeassistant.setup
Source: setup.py:215
First occurred: 06:21:21 (1 occurrences)
Last logged: 06:21:21

Setup failed for custom integration spaclient: Unable to import component: No module named 'homeassistant.util.temperature'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 215, in _async_setup_component
    component = integration.get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 816, in get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_components/spaclient/__init__.py", line 19, in <module>
    from .spaclient import spaclient
  File "/config/custom_components/spaclient/spaclient.py", line 9, in <module>
    from homeassistant.util.temperature import convert as convert_temperature
ModuleNotFoundError: No module named 'homeassistant.util.temperature'"

plmilord, 2023-11-04:
I had the same problem with another user yesterday...

Make sure you have a /spaclient folder in /custom_components on your Home Assistant installation. Otherwise:
1-Copy the /spaclient in /custom_components
2-Do a complete reboot of your HA
3-Proceed to uninstall Spa Client
4-Do a complete reboot of your HA
5-Install Spa Client again (manually or via HACS)

Let me know the results...or the recipe that worked for you...I'll add this as a comment on version v2.8.

IkaSGC, 2023-11-04:
The reinstallation worked. 
I'll ask.... Wouldn't it be possible to somehow add circulating pump control ??

IkaSGC, 2023-11-04:
Oh sorry... Recipe: I've removed the integration. In HACS had SpaClient reinstalled, it's strange that it wasn't installed. And then I added the integration again.

closed 2023-11-15 by plmilord