plmilord/Hass.io-custom-component-spaclient #43: Deprecated constanst from the core 2025-01

berkavil, 2024-01-18:
Hallo, since the last core update I started to receive these warnings:

Logger: homeassistant.const
Source: helpers/deprecation.py:205
First occurred: 12:17:18 (10 occurrences)
Last logged: 12:17:20

TEMP_CELSIUS was used from spaclient, this is a deprecated constant which will be removed in HA Core 2025.1. Use UnitOfTemperature.CELSIUS instead, please create a bug report at https://github.com/plmilord/Hass.io-custom-component-spaclient/issues
TEMP_FAHRENHEIT was used from spaclient, this is a deprecated constant which will be removed in HA Core 2025.1. Use UnitOfTemperature.FAHRENHEIT instead, please create a bug report at https://github.com/plmilord/Hass.io-custom-component-spaclient/issues

I hope we will be stil able to use this integration after the update in Jan-2025?
thank you.
Willy

berkavil, 2024-01-18:
... and some more here (these are different warnings shown in my logs, so I am sorry for sendin them separated here...:
DEVICE_CLASS_CONNECTIVITY was used from spaclient, this is a deprecated constant which will be removed in HA Core 2025.1. Use BinarySensorDeviceClass.CONNECTIVITY instead, please create a bug report at https://github.com/plmilord/Hass.io-custom-component-spaclient/issues

STATE_CLASS_TOTAL_INCREASING was used from sems, this is a deprecated constant which will be removed in HA Core 2025.1. Use SensorStateClass.TOTAL_INCREASING instead, please create a bug report at https://github.com/TimSoethout/goodwe-sems-home-assistant/issues

thanks for any information about solving this.
Willy

plmilord, 2024-01-21:
Thanks @berkavil... I'm aware of these deprecated items. I'll be back soon with an update.

berkavil, 2024-01-21:
Thank you for a quick answer @plmilord ...
All working so far except the warning msgs. Thanks indeed for this marvelous spaclient integration. :-)

cappadanna, 2024-02-05:
Will the BinarySensorDeviceClass.CONNECTIVITY  be used in future update instead to see the wifi module status? It reports unavailable right now

plmilord, 2024-02-05:
@cappadanna,

As noted in v2.81, I removed the SpaGateway entity because I now manage entity availability. In fact, the binary sensor “bwa Wi-Fi Module” never worked well. If you have automations/scripts depending on whether the Wi-Fi module is online or not, you can now validate the availability of any other entity to carry out the monitoring work (personally, I use the "Spa Thermostat" entity).

berkavil, 2024-02-07:
> Deprecated stuff now been cleared. Have a look at [v2.81](https://github.com/plmilord/Hass.io-custom-component-spaclient/releases/tag/v2.81).
> 
> Let me know if this works correctly for you...

Hi @plmilord thank you indeed, I installed the 2.81 today and so far no warning signs seen. Hope all works as before :-)
thank you for clearing this.
W.

closed 2024-02-08 by plmilord