plmilord/Hass.io-custom-component-spaclient #44: Filter Run Time Says It Ends At 2 am but that's actual run time.

monsieurlatte, 2024-02-07:
Just installed this and thank you so much for this!  However one bug is that the filter 1 and 2 start times are correct but they say the end time is 2am but it's just 2 hour runtime so I think we're getting the data correctly but it's not for end time but run time.  Thoughts?

monsieurlatte, 2024-02-07:
Yep, I changed it to run for 2 hours and 15 and now the end time says 2:15 am.

Jieper001, 2024-02-07:
Same here, that 'AM' shouldn't be there.

plmilord, 2024-02-08:
:-) Good eyes! I haven't touched the code on this side... probably an evolution in the Time entity. I'll give it a look!

plmilord, 2024-02-08:
Please review your Profile settings in Home Assistant. The time format is set to Auto by default...and in English the default is AM/PM. You can change the format to omit AM/PM.

If I find a way to force it, I will adjust the component.

[image]

monsieurlatte, 2024-02-08:
Yeah it's not from that, it's that it's a runtime not an end time ?  appreciate the help.
[image]

plmilord, 2024-02-08:
Yes it's a runtime.

The Home Assistant "Time" entity displays it like this... If you want to eliminate AM/PM from these fields, adjust your profile settings.

My filtration cycle 1 starts at 08:00 and lasts 04:00 hours
My filtration cycle 2 starts at 16:00 and lasts 16:00 hours

Without AM/PM, no confusion...

[image]

monsieurlatte, 2024-02-08:
got it, thanks.  This repo has it the other way if that helps you at all.  https://github.com/garbled1/balboa_homeassistan

again no pressure as I am grateful for this one as it is :)

plmilord, 2024-02-08:
I tried to  to represent the App **Coast Spas** as faithfully as possible:

[image]

You can also find useful attributes under "Filter Cycle x Status" binary sensor:

[image]

monsieurlatte, 2024-02-08:
Yeah, I like having my other stuff in that method so what I did instead was just create a helper sensor from the time using this template.

"{{ states('time.filter_cycle_1_runs') }}"

closed 2024-02-08 by plmilord

Jieper001, 2024-02-10:
I just updated my profile settings to show 24h format and all is good.  Thanks again for a great project (my SpaTouch touchscreen died recently and I can still do all I need via HA now - top!).

Profile settings: ip-addressHA/profile