plmilord/Hass.io-custom-component-spaclient #45: Upgrade to 2.81 creates a new unit with new entities

markuswesterberg, 2024-02-10:
This upgrade needs to be revised. When upgrading it creates a second, duplicate unit with new entities named "....._2" and the original unit becomes unavailable.

Jieper001, 2024-02-10:
I can confirm the same thing happened to me.  Not a major issue though.

plmilord, 2024-02-10:
What version did you upgrade from? It's as if you were using a version older than 2.8. As noted in the 2.8 version notes, you need to uninstall the old version and reinstall this new one.

No _2 here...:

[image]

markuswesterberg, 2024-02-10:
Ah! OK. My bad, I did upgrade from an older version. Will restore back-up and read the notes before upgrading, thanks!!

plmilord, 2024-02-10:
Make sure to uninstall everything from Spa Client first, reboot, then install the latest version... this should be back under control for you!

closed 2024-02-10 by plmilord