plmilord/Hass.io-custom-component-spaclient #47: Wi-Fi bwa where is ?

Percherie, 2024-02-13:
Hello 

The wifi module is no longer available. What is recommended to check that the device is correctly connected? 
The binary sensor was perfect for this. Currently following a firmware change I can no longer query the router and I'm looking for a simple entity to query, without having to add new integrations.

Since the update I've been disconnected from the wifi several times an hour and I need to check that it's online before executing scripts.

Traduit avec DeepL https://www.deepl.com/app/?utm_source=android&utm_medium=app&utm_campaign=share-translation

plmilord, 2024-02-13:
Salut @Percherie,

L'entité WiFi devenait inutile car j'ai revu le fonctionnement de la disponibilité des entités (et ça fct super bien maintenant).

Pour ma part, je ne fais que valider la disponibilité d'une des entités du spa (dans mon cas, "Spa Thermostat"). Je me suis fait une automatisation pour me texter en cas de défaillance prolongée.

[image]

Percherie, 2024-02-13:
Merci pour le retour rapide

Je finalise un automate qui me permet de gérer la qualité de l'eau grace à la sonde Flipr qu'on peut intégrer au HA
Es? t-ce que t'intéresserait que je te partage

plmilord, 2024-02-13:
J'aimerais bien voir ça! Je vais fermer ce problème soulevé... Nous pouvons transporter ce sujet (sonde Flipr) dans l'onglet Discussions. Merci à toi!

closed 2024-02-13 by plmilord