plmilord/Hass.io-custom-component-spaclient #50: Unable to select temperature 37.5°C

Percherie, 2024-04-02:
How do you do? 
When very hot mode is selected and I want to heat to 37.5, the temperature is automatically changed to 38°C. 
I've tested with the official balboa application and I don't have this type of problem. 

Is this a problem with the HA thermostat tile or is it a problem with the integration?

DBtun, 2024-04-19:
I have the same problem. I can set 36.5 and 38.5, but not 37.5 degrees celcius. Could this be related to 37.5 degrees is close to 100 degrees farenheit? Or maybe a round off problem?
Using the balboa app works fine to set 37.5.

plmilord, 2024-04-20:
I ran into this issue too...I'll take a look!

plmilord, 2024-04-20:
The behavior also happen while set temp at 27.5°C , 32.5°C and 37.5°C...

Percherie, 2024-04-20:
I hadn't noticed that. I can confirm that this is also the case for me

plmilord, 2024-04-21:
Please check the latest version [v2.83](https://github.com/plmilord/Hass.io-custom-component-spaclient/releases/tag/v2.83)... This should resolve the issue.

Let me know if this works for you.

Thanks!

Percherie, 2024-04-21:
Great job, this solves the problem perfectly

Thanks for your responsiveness

Ps: I haven't forgotten you for the "On-demand temperature management" script share... I've renamed the entities in French, do you speak French or do I have to translate the English entities I've used ???

DBtun, 2024-04-21:
Yes, problem solved. Great and quick work!

plmilord, 2024-04-21:
No problem guys!

@Percherie... pour le français, c'est bon, j'habite le Québec et ma langue maternelle est le français!

closed 2024-04-21 by plmilord

plmilord, 2024-08-10:
Oui j'ai vue... bcp de travail là-dedans! Je n'ai malheureusement pas essayé tout ça. Sera très certainement utile pour la communauté!