plmilord/Hass.io-custom-component-spaclient #54: Future request: Hold mode

fatuuse, 2024-08-03:
I would love if the **Hold mode** had a switch in the component.
`0x3C 	Hold Mode`

Thanks for your work with the custom component!

plmilord, 2024-08-10:
Thanks for letting me know... seems like there's some new development in there... I'll try these new switches:

0x01 | Normal Operation (exit Priming Mode, etc.)
0x03 | Clear notification (reminders, etc.)
0x1D | Soak Mode (All pumps off)
0x3C | Hold Mode

GroteGehaktBal, 2024-08-17:
I would really like the hold mode feature. I know that the there is already a plugin for homebridge that has a hold mode button.

[https://github.com/vincedarley/homebridge-plugin-bwaspa/blob/master/src/holdSwitchAccessory.ts](url)

plmilord, 2024-09-02:
Take a look at version [v3.2](https://github.com/plmilord/Hass.io-custom-component-spaclient/releases/tag/v3.2). `Hold Mode` and `Standby Mode` are now implemented.

Let me know if this works for you...

fatuuse, 2024-09-07:
Hold mode works as expected! Thanks!!

closed 2024-09-07 by fatuuse