plmilord/Hass.io-custom-component-spaclient #55: Feature request: Timesync

rploeg, 2024-08-04:
Would be great to set a timesync of the controller. Sometimes my power goes down and then the spa control don’t have the right time. In the app I have to hit the sync time button to solve my issues. Would be great to set this also via HA.

plmilord, 2024-08-10:
Hello @rploeg,

When you navigate to the Spa Client integration, you can configure time synchronization if you wish...

The synchronization mechanism will run every 24 hours after the first synchronization (first enabled or HA reboot).

[image]
[image]

Let me know if this works for you as per your needs!

rploeg, 2024-08-10:
That is great and will do the trick! Sorry didn't see the option before. Thanks for your great work!

image.png (view on web)<https://github.com/user-attachments/assets/1809f657-82dc-4e46-bac1-22ecd314baf2>
image.png (view on web)<https://github.com/user-attachments/assets/ede7d692-98ad-4160-b1ac-66c178cb960c>

You are receiving this email because you were mentioned.

closed 2024-08-10 by plmilord