plmilord/Hass.io-custom-component-spaclient #57: Modity filter cycles

javo222, 2024-08-27:
Hi,
I have a plugin that finds the electricty's cheapest spot price everyday and I would like to combine it with the spa client to update the filter_cycle_1_begins and filter_cycle_1_runs everyday. For instance at 2pm I check how the price looks like for the next day and adjust the filter time.

I didn't find a way to update both entities, is this possible? Or can it be added to the wishlist?
Thanks

berkavil, 2024-09-12:
Hi, you can create the automation and turn on-off these filters via HA script.

fatuuse, 2024-12-03:
Like this:

```
- alias: Pool update price
  id: "pool_price"
  trigger:
  - platform: time
    at: "23:50"
  action:
  - variables:
      lowest_x: 6
      tider: >-
        {%- set l=(state_attr('sensor.nordpool_kwh_se3_sek_3_095', 'raw_tomorrow') | sort(attribute='value'))[0:lowest_x] | sort(attribute='start') %}
        {%- set ns = namespace() %}{% set ns.start1 = '' %}{% set ns.end1 = '' %}{% set ns.start2 = '' %}{% set ns.end2 = '' %}
        {%- for i in l %}
        {%- if ns.start1 == '' %}{% set ns.start1 = i.start.hour %}
        {%- elif i.start.hour != ns.end1 and ns.start2 == '' %}
        {%- set ns.start2 = i.start.hour %}
        {%- endif %}
        {%- if i.end.hour == (int(ns.end1,0)+1) or ns.end1 == '' %}
        {%- set ns.end1 = i.end.hour %}
        {%- elif i.end.hour == (int(ns.end2,0)+1) or ns.end2 == '' %}
        {%- set ns.end2 = i.end.hour %}
        {%- endif %}
        {%- endfor %}
        {{ [ns.start1, ns.end1, ns.start2, ns.end2] }}
  - wait_template: "{{ has_value('time.filter_cycle_1_begins') }}"
    timeout: "03:00"
    continue_on_timeout: false
    alias: "wait to be online"
  - service: time.set_value
    target:
      entity_id: time.filter_cycle_1_begins
    data:
      time: "{{ tider[0] }}:00"
  - service: time.set_value
    target:
      entity_id: time.filter_cycle_1_runs
    data:
      time: "{{ int(tider[1],0) - int(tider[0],0) }}:00"
  - if:
    - condition: template
      value_template: "{{ tider[2] is number and tider[3] is number }}"
    then:
    - service: time.set_value
      target:
        entity_id: time.filter_cycle_2_begins
      data:
        time: "{{ tider[2] }}:00"
    - service: time.set_value
      target:
        entity_id: time.filter_cycle_2_runs
      data:
        time: "{{ int(tider[3] | replace('0','24'),0) - int(tider[2],0) }}:00"
    - service: switch.turn_on
      target:
        entity_id: switch.filter_cycle_2
    else:
    - service: switch.turn_off
      target:
        entity_id: switch.filter_cycle_2
```

sensor.nordpool_kwh_se3_sek_3_095
```
raw_today:
  - start: "2024-12-03T00:00:00+01:00"
    end: "2024-12-03T01:00:00+01:00"
    value: 0.428
  - start: "2024-12-03T01:00:00+01:00"
    end: "2024-12-03T02:00:00+01:00"
    value: 0.427
  - start: "2024-12-03T02:00:00+01:00"
    end: "2024-12-03T03:00:00+01:00"
    value: 0.202
...
```