plmilord/Hass.io-custom-component-spaclient #63: BP2100G1 and TP700 top side

stefancn, 2024-10-22:
Hi !  Nice integration ... but I've some discrepancies ...

Filtercycles setting:

In the TP700 top. the filtercycles is set in 15 minutes increment ,  if I set to 070:00 and 07:18 the starttime is ok on the TP700 but the stop time is garbage ...   So it would be nice to have a additional step setting to the filtercycles.

Last Known fault.

The fault "Sensors are out of sync" is not shown in the app.

Regards ...

Stefan

plmilord, 2024-10-22:
I'll take a look to get that 15 minute increment!

Regarding the "sensors are not in sync" message, could you give me the "self.fault_log_msg_code = ___" print in your HA logs. I'll add it to the component dictionary.

Thanks.

stefancn, 2024-10-23:
Hi !   and thanks for your prompt reply, much appreciated,  now I saw the "sensors not in sync" message so it is announced.

Quest 1 .  Is it possible to add a button to clear the "alarm" eg. emulate the "X" button on the TP700. 
Quest 2.   Is it possible to add a "clear log" button for the "faults" ....

Can you also fix so when I set the Filtercycle 1 to be active 24 hour, I set the begin to "00:00" and runs for "00:00" the TP700 sets running time to "24 hour" , but when I set the same into the Spa Client the TP700 shows "Begins" to 00:00 and the "Runs" to 00:15.

... and last where do I find the HA logs for the Spa Client ... is it embedded in the systemslog or ...

Again thanks for your prompt reply ...

// Stefan

plmilord, 2024-10-26:
A lot of to does!

For the buttons, if you clear the log history on your TP700 (press the X button) and then do a HA reboot, does your Spa Client still show the last error (I don't have my spa to test this)? Probably yes, since the history is stored locally on the TP700 and in the HA Spa Client component. If this works as expected... then I will imagine a button to acknowledge the last error, to easily distinguish the arrival of a new error!

For logs, after a HA reboot, just go to: Settings/System/Logs... then click on "Load full logs". You will see a full list of Spa Client variables! What is the "self.fault_log_msg_code = ___" displayed?

stefancn, 2024-10-27:
Thanks ... I understand that you have a lot to do but can you at least make a small change to accept Filter start, and running times of : 00:00 so I can get the spa to run the filter cycle of 24 hours.
What  IDE, and type, version of the Pyrhon environment do you use.  As an old C++ programmer in industrial systems, the Python and ESP32 is completely new to me.

Thanks in advance ...

//Stefan

plmilord, 2024-11-02:
I only use Notepad++ to edit the code. Since it's 100% Python, no need to compile anything! You can use a more code-friendly development tool like VScode. As you wish!

As soon as I have time, I will examine your needs in more detail... As my spa is down for this winter, it will rather be for next summer.