plmilord/Hass.io-custom-component-spaclient #65: Problem with unresponsive integration

labbs1, 2024-11-26:
Hi,

I have had problems with this integration going dark from time to time and the last 2-3 months it has been even worse. Since I am not able to reload it, the only thing that gets it going is to restart HA. After that it works for tops a day or so. Haven't enable debugging but the error message in HA is:

`Logger: homeassistant.helpers.service
Källa: helpers/service.py:302
Inträffade först: 25 november 2024 kl. 23:05:00 (8 händelser)
Senast loggade: 16:44:10

Referenced entities climate.spa_thermostat are missing or not currently available`

I don't know how to troubleshoot this but can share what ever needed and on an additional note, I have automations setting the spa in rest mode from time to time in order to save energy . Could this be an issue? I have a feeling it does not come back after a while in REST mode?

Thanks!

Jieper001, 2024-11-26:
Chipping in: this not related to the fact that the Balboa Wifi module does go down from time to time for exactly 60mins?  (not at home so can't look into more detail)

labbs1, 2024-11-27:
> Chipping in: this not related to the fact that the Balboa Wifi module does go down from time to time for exactly 60mins? (not at home so can't look into more detail)

Intreseting but I have not seen or noticed this behaviour. In my case it is more; once it looses connection with HA it does not come back until HA is restarted but I can still access the spa via Balboa's app

plmilord, 2024-11-27:
It looks like you're not up to date!? Make sure you have [v3.2](https://github.com/plmilord/Hass.io-custom-component-spaclient/releases/tag/v3.2) installed

labbs1, 2024-11-28:
I am running on 3.2 according to the manifest. 

I restarted HA yesterday around noon and at 02:03 this night it became unresponsive again

monsieurlatte, 2024-12-01:
I’m having something similar. Mi e won’t connect at all now. I even rebooted my hot tub system.  I can connect via cloud with no issue.  Maybe a firmware update on the WiFi module is causing the issues?

monsieurlatte, 2024-12-01:
FWIW I used the built in balboa one from HA and it works. So has to be something with this fork.

plmilord, 2024-12-06:
At this point I'm no help...my spa is out of commission for this winter! Will give this a look asap next spring!

Can someone please proceed with a complete re-install (uninstall, reboot, delete \\homeassistant\config\custom_components\spaclient folder, make sure there are no more traces of spaclient entities, reboot, install the latest version [v3.2](https://github.com/plmilord/Hass.io-custom-component-spaclient/releases/tag/v3.2)).

There were some things that were supposed to arrive in late 2024 in the Home Assistant climate entity reported by @labbs1... things that have been fixed for a while now! That's why I'm attempting a full reinstall of this custom component with one of you!

Let me know!